benaclejames/VRCFaceTracking #179: Face tracking completely bricks itself for Vive Pro Eye

SakuroJason, 2023-11-24:
Hey, my experience with VRC Face Tracking was very good even though it crashed a lot but now my lip tracker does some weird things every time I use the headset. It's different every time I use it but it never tracks my face like it used to. I bought a new lip tracker to see if that is the problem but if this problem presists it has to do with VRC Face Tracking.

Kind regards

Jason

RantiMess, 2023-11-25:
Have you tried replacing the headset cable?

SakuroJason, 2023-11-25:
Yes it didn’t help

benaclejames, 2023-11-29:
Is this resolved? If you need any more debugging or help, I'd highly recommend asking in [the discord](https://discord,gg/vrcft) where people who may have had the same issues can help you.

SakuroJason, 2023-11-30:
I visited a group instance on VRChat from VRCFaceTracking and a user told me to disable automatic calibration. I don’t understand why it turns itself on and breaks my face tracking but I rather have a reinitialize button so if a device disconnects and reconnects, I don’t have to restart the whole program.

closed 2023-12-24 by benaclejames