benaclejames/VRCFaceTracking #182: Crashing with Quest Pro

Kiwidogg, 2023-11-29:
I'm having issues with my pc whenever I turn on vrcft, eveyrtime I open and press "toggle tracking", vrc will track and work fine for 5-10 minutes or so, but then my vr and desktop freeze. After a few seconds my monitors will go black and turn back on, and I'll be greeted by an error message on SteamVR and wont be able to reconnect to link until I reset my pc. I have used link and airlink and have searched up several videos and articles and I can't find a solution. I have tried using both methods for oculuslinks (ALXR Local and Quest Pro OpenXR modules) and neither has worked. I have done tests to see what exactly is causing this and I've realized that it only happens whenever I have vrcft tracking toggled and SteamVR running at the same time. I also believe it's something with my graphics drivers since one time when my pc did this crash, only one DP was working (no HDMI) until I re-enabled my drivers, which I assume were disabled as a result of this crash.

Here's my setup
16gb 3600mhz ram
AMD cpu and gpu (I saw an issue somewhere similar-ish about an AMD GPU)
500W psu
3 vive trackers
2 base stations

It's not something with SteamVR from what I've observed, I've tried using my Index and it works flawlessly when I use SteamVR with it, so I'm really out of ideas.

I've tried reinstalling, manually downloading vrcft and the modules, and reinstalling SteamVR , and nothing has worked (all with the help of youtube videos and/or doc linked on the github page). I feel like it's something causing conflict between SteamVR and vrcft but I'm no expert, is there anything I can do? I mainly got the quest pro for the face tracking so I hope soo. Please and thank youu, I really hope it wasn't a waste of money:(

benaclejames, 2023-11-29:
This sounds like a hardware issue. Have you looked or tried asking the same question in [the discord](https://discord.gg/vrcft)?

kenthinson, 2023-12-07:
> I'm having issues with my pc whenever I turn on vrcft, eveyrtime I open and press "toggle tracking", vrc will track and work fine for 5-10 minutes or so, but then my vr and desktop freeze. After a [...]

Quest Pro here. No issues with SteamLink. Can use tracking all day long. Try it out instead. You will have to install the module from github

benaclejames, 2023-12-24:
@Kiwidogg is this resolved?

closed 2024-02-12 by benaclejames