benaclejames/VRCFaceTracking #197: [Feature Request] Linux Support?

muskwasis, 2024-02-27:
I am able to use 99 percent of the features in VRChat (dj live streams are the only thing broken), and FBT through slimevr on Linux. Right now I use a Windows VM running VRCFT and sending the OSC data back and forth with firewall rules, but ideally would like to run the app directly on Linux. I'd like to dig deeper in to this, just checking in case someone already has?

Mhowser, 2024-05-19:
Looking for this as well, I'm using the Steam Deck with Steam OS which is based on Arch.

ShayBox, 2024-07-06:
Also looking for this, it's in C# so should is very possible.

kruemmelbande, 2024-07-06:
Also consider looking into https://github.com/galister/oscavmgr. It does basically the same thing and runs on linux

ShayBox, 2024-07-06:
Thank you :smile:

dfgHiatus, 2025-07-17:
Better late than never lol https://github.com/dfgHiatus/VRCFaceTracking.Avalonia