benaclejames/VRCFaceTracking #204: VRCFT wont run because I have no way to launch it.

Paperreadmen, 2024-04-15:
So I downloaded VRCFaceTracking_x64 installer. I run the installer and it says "install" or "launch". I run VRCFT with that launch button each time I wanna run it. But... for some reason VRCFT changed the installer to no longer have the "launch" option! so... I can run it.

I search everywhere on my computer for any exe or SOMETHING to run it. I searched "VRCFaceTracking_x64" "VRCFacetracking" "VRCFT" "VRCFaceTracking.exe"... Nothing shows a way to run the damn thing.

Apparantly, a friend told me thats not how they run it, they just use a shortcut to what I assume is the exe... But I cant find it!! I tried to uninstall VRCFT, turns out theres not an uninstall feature I can find... I tried deleting VRCFT files everywhere I can find them, they wont delete. I dont know why either, wont let me. I tried just hitting "install" again, says I cant and theres already a package for it that I have to remove. I found that package, deleted it, says I STILL have it. I'mma bout to pull my hair out.

PanDamien26, 2024-04-15:
VRCFT is a UWP Type app and doesn't install like a regular old .exe
It should normally be listed in your start menu.
Location is C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.1.1.0_x64__d7rcq4vxghz0r\VRCFaceTracking.exe

dfgHiatus, 2025-07-24:
To resolve this, please download the latest release of VRCFaceTracking from [Steam](https://store.steampowered.com/app/3329480/VRCFaceTracking/).

closed 2025-07-24 by dfgHiatus