benaclejames/VRCFaceTracking #206: Everything is running, but no data is being sent and FT module is off but connected.

SyntheticMelody, 2024-04-28:
[image]

[image]

[image]

[image]

[image]

Anyone know a fix to this? I am running a vive pro, index controllers, fbt and the vive face tracker. It was working perfectly fine last night, but now it's IR ace sensors are completely turned off despite being connected.

ThatGuyThimo, 2024-10-08:
Generally these issues are better asked in the discord group. 

It looks like no modules are being loaded by the VRCFT App make sure you have the SRanipalTrackingModule installed in the module Registry tab, also make sure you have SRanipal Runtime installed on your computer.

hope this helps

dfgHiatus, 2025-07-24:
Closing this issue due to inactivity.

closed 2025-07-24 by dfgHiatus