benaclejames/VRCFaceTracking #207: "Sending" doesn't work but "listening" does

SparkYouTube, 2024-05-18:
I have a Quest Pro, using SteamLink
I promise you, I have done every tutorial on the internet and nothing has fixed this
Nothing works

Yes, I have developer mode on
Yes, face and eye tracking is on
Yes, I have reset the avatar, reset OSC, reset the config
Yes, I have tried avatars made by VRCFT for facial tracking test

The code constantly says 
[SteamLinkVRCFTModule] Information Exception A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.

[VRCFaceTracking doesn't work.txt](https://github.com/benaclejames/VRCFaceTracking/files/15358973/VRCFaceTracking.doesn.t.work.txt)

InconsolableCellist, 2024-05-18:
The error's coming from connection code in the SteamLinkVRCFTModule (https://github.com/ykeara/LinkFT) while attempting to connect to OSC. Did you also check the settings on "OSC Output Port" and similar, which you can see on the linked README.md?

SparkYouTube, 2024-05-18:
Where is the OSC Output Port?

InconsolableCellist, 2024-05-18:
[image]

SparkYouTube, 2024-05-18:
Just saw it after deleting that message. I am small brain X3

SparkYouTube, 2024-05-18:
You're a fucking genius
God bless you

closed 2024-05-18 by SparkYouTube