benaclejames/VRCFaceTracking #218: Quest Module werden nicht geladen

TheD4rknet, 2024-06-30:
Die App kann keine Connection zu meiner Quest Pro aufbauen bzw. keine Verbindung zu den Modulen. Ich weiss nicht wo das Problem liegt, hab schon einiges probiert, aber leider ohne Erfolg. 

Über eure Hilfe und Unterstützung wäre ich sehr dankbar.

Chickenbreadlp, 2024-06-30:
Brudi, du weißt schon das eigentlich jeder der an diesem Repo arbeitet nur English kann, oder?

For thos who do not speak german, here's a translation:
> The app cannot establish a connection to my Quest Pro or a connection to the modules. I don't know what the problem is, I've already tried a few things, but unfortunately without success.
> 
> I would be very grateful for your help and support.

TheD4rknet, 2024-07-09:
> Brudi, du weißt schon das eigentlich jeder der an diesem Repo arbeitet nur English kann, oder?
>
> For thos who do not speak german, here's a translation:

TheD4rknet, 2024-07-09:
Hab ich nicht dran gedacht, danke dir.

closed 2024-09-25 by benaclejames