benaclejames/VRCFaceTracking #232: Changing OSC ip Address cause application to hang and crash

ThatGuyThimo, 2024-10-06:
When using vrcfacetracking, trying to change the ip to my other computer in the OSC address causes the application to freeze/crash

VRCFT V: 5.3.3.0
Logs didn't say anything about the OSC

AlexisEvo, 2024-10-06:
Dumping some info from when I last looked into this on Discord: currently, VRCFT both listens and sends to the IP address configured in the UI. IIRC the crash occurs because it can't bind to an address that isn't on the local machine. Fixing this for the non-oscquery implementation would require different UI config option for IP addresses for listening/sending.

The second part of the issue is that OSCQuery hardcodes the address it advertises to "127.0.0.1", so if VRC is running on a different computer, it won't be able to send/receive to VRCFT. This seems intentional due to [this code/comment](https://github.com/benaclejames/VRCFaceTracking/blob/25f7e9548505978be648e8ecf3a507afc14103c3/VRCFaceTracking.Core/mDNS/MulticastDnsService.cs#L233-L238), but this breaks setups where VRCFT is running on a separate PC -- especially so for someone trying to use standalone VRC.

ThatGuyThimo, 2024-10-06:
> Dumping some info from when I last looked into this on Discord: currently, VRCFT both listens and sends to the IP address configured in the UI. IIRC the crash occurs because it can't bind to an [...]

i don't think that this is an issue since you are able to change the ip in launch settings for vrchat like stated [here](https://docs.vrchat.com/docs/osc-overview), and the ports / i do think that it is an binding issue on the part of the application, i have just build a version localy and confirmed two issues with the ip field

if you clear the ip field than the application will crash and next time you launch it the same thing will happen, i looked into the appdata/roaming/VRCFaceTracking folder for the settings json and there the ip would be blank, if you replace that with a valid ip the app would work correctly again, so something is going wrong when validating the saved address and when setting it to blank it doesn't correct its self to the default 127.0.0.1,

i have also been able to confirm that the app hangs when setting it to a remote ip instead of straight up crashing

Toys0125, 2024-12-20:
I think i can also add changing the listen port to 0 to allow OCS Query only the application hangs on startup/launch if there's no OCS Query service detected. Example if VRChat wasn't opened before the application was launched.