benaclejames/VRCFaceTracking #237: Is there complete list of OSC Parameters provided by / read by VRC Face Tracking?

anatawa12, 2024-10-30:
Hello. I'm the author of [Avatar Optimizer](https://an12.net/aao).

Is there any documentation or source code location that provides all parameters VRCFaceTracking provide to / read from Avatar?

I'm planned to implement more optimization for Animator Controller in the future release of Avatar Optimizer assuming parameters are not animated.
However, this optimization would break compatibility with OSC Tools, so Avatar Optimizer is going to have system to declare parameter information about OSC Tools since next release, and bundle information for popular tools like VRCFaceTracking.

Therefore, I want to know all parameters VRCFaceTracking provide to / read from Avatar.

I found https://docs.vrcft.io/docs/tutorial-avatars/tutorial-avatars-extras/parameters but this says it's incomplete, and there are no information about which parameters are read by VRC Face Tracking.
> There are many parameters listed here, and some that are undocumented to help users save performance (such as 100% optimal parameter replacements for some parameters). 

I found https://github.com/benaclejames/VRCFaceTracking/blob/master/VRCFaceTracking.Core/UnifiedTracking.cs#L37-L46 and it looks complete but I could not trust my assumption so I asked here.
And this also doesn't have information about which parameters are read by VRC Face Tracking.

lonelyicer, 2024-11-03:
```
v2/EyeSquintRight
v2/EyeSquintRight
v2/EyeSquintRightNegative
v2/EyeSquintLeft
v2/EyeSquintLeft
v2/EyeSquintLeftNegative
v2/EyeWideRight
v2/EyeWideRight
v2/EyeWideRightNegative
v2/EyeWideLeft
v2/EyeWideLeft
v2/EyeWideLeftNegative
v2/BrowPinchRight
v2/BrowPinchRight
v2/BrowPinchRightNegative
v2/BrowPinchLeft
v2/BrowPinchLeft
v2/BrowPinchLeftNegative
v2/BrowLowererRight
v2/BrowLowererRight
v2/BrowLowererRightNegative
v2/BrowLowererLeft
v2/BrowLowererLeft
v2/BrowLowererLeftNegative
v2/BrowInnerUpRight
v2/BrowInnerUpRight
v2/BrowInnerUpRightNegative
v2/BrowInnerUpLeft
v2/BrowInnerUpLeft
v2/BrowInnerUpLeftNegative
v2/BrowOuterUpRight
v2/BrowOuterUpRight
v2/BrowOuterUpRightNegative
v2/BrowOuterUpLeft
v2/BrowOuterUpLeft
v2/BrowOuterUpLeftNegative
v2/NasalDilationRight
v2/NasalDilationRight
v2/NasalDilationRightNegative
v2/NasalDilationLeft
v2/NasalDilationLeft
v2/NasalDilationLeftNegative
v2/NasalConstrictRight
v2/NasalConstrictRight
v2/NasalConstrictRightNegative
v2/NasalConstrictLeft
v2/NasalConstrictLeft
v2/NasalConstrictLeftNegative
v2/CheekSquintRight
v2/CheekSquintRight
v2/CheekSquintRightNegative
v2/CheekSquintLeft
v2/CheekSquintLeft
v2/CheekSquintLeftNegative
v2/CheekPuffRight
v2/CheekPuffRight
v2/CheekPuffRightNegative
v2/CheekPuffLeft
v2/CheekPuffLeft
v2/CheekPuffLeftNegative
v2/CheekSuckRight
v2/CheekSuckRight
v2/CheekSuckRightNegative
v2/CheekSuckLeft
v2/CheekSuckLeft
v2/CheekSuckLeftNegative
v2/JawOpen
v2/JawOpen
v2/JawOpenNegative
[... 1008 lines cut ...]
MouthTightener
MouthTightenerNegative
MouthTightenerLeft
MouthTightenerLeft
MouthTightenerLeftNegative
MouthTightenerRight
MouthTightenerRight
MouthTightenerRightNegative
NoseSneer
NoseSneer
NoseSneerNegative
NoseSneerLeft
NoseSneerLeft
NoseSneerLeftNegative
NoseSneerRight
NoseSneerRight
NoseSneerRightNegative
BrowDownLeft
BrowDownLeft
BrowDownLeftNegative
BrowDownRight
BrowDownRight
BrowDownRightNegative
```
That should be all.

lonelyicer, 2024-11-03:
```
foreach (var parameter in UnifiedTracking.AllParameters_v2.Concat(UnifiedTracking.AllParameters_v1)
                 .ToArray())
{
    var paramNames = parameter.GetParamNames();
    foreach (var (paramName, paramLiteral) in paramNames)
    {
        _logger.LogInformation($"{paramName}");
    }
}
```
I used this method to output them, so there may be duplicates in there.
This method comes from here: https://github.com/benaclejames/VRCFaceTracking/blob/master/VRCFaceTracking.Core/OscQueryConfigParser.cs#L55-L59

benaclejames, 2024-11-03:
Don't forget about binary parameters as well! Any of the above parameters can be prefixed by a number of power of 2. These are generated at runtime and so might need special handling to detect. We use a regex pattern internally

anatawa12, 2024-11-04:
Thank you for extracting list.

> Don't forget about binary parameters as well!

Thank you for note that. I didn't know about that.
I'll add support for regex pattern for better support.
You mean something like `v2/EyeSquintRight1` are valid.

It looks lonelyicer's list is about parameters VRCFT sends to VRChat, but is there any parameters VRCFT reads from VRChat?