benaclejames/VRCFaceTracking #245: Certificate on older versions.

skunkmyrddyn, 2024-12-10:
I'm trying to debug an issue with an avatar and having eye tracking work and disable itself.  Other users have indicated that using 5.1.1.0 worked for them.  However, the certificate alongside 5.1.1.0 is no longer valid, and the previous version won't install.

I tried to download the source code and compile it, but it requires "deployment" instead of allowing to run locally and complains a lot about app manifest issues.   

Sorry if this sounds like 2 issues but how do I either: 
Install 5.1.1.0 from the download or 
Compile 5.1.1.0 for local running without needing to deploy it to the MS Store.

AlexisEvo, 2024-12-10:
Download the .msix file from the releases page, extract it as a .zip file, then run the executable from there.

skunkmyrddyn, 2024-12-10:
This worked.  Also fixed the bug I was having.   Makes me wonder what changed between 5.1.1.0 and the current 5.3.x that causes the eye tracking to spontaneously switch to random wandering.

benaclejames, 2024-12-10:
Random wandering is not a functionality implemented from within VRCFT. This is likely caused by a pause in eye tracking data being sent to vrchat's native eye tracking system. I've seen this caused by modules being unable to get/send eye tracking data to vrchat for extended periods of time.

As for why the application is difficult to install on older versions: Microsoft's app installer format enforces a valid code signing certificate. Older versions of vrcft utilize a now-expired certificate and, as such, are prevented from installing in the normal manner.

Hope this helps.

skunkmyrddyn, 2024-12-11:
Thanks for the replies. 
It does help explain why the 5.1.1.0 won't install "as-is" and needs to be unzipped and run manually.
It doesn't explain why the avatar reacts differently under 5.1.1.0 and the current latest version.   I understand that the wandering eye movement is internal to VRC; but that's what it looks like when I run the latest VRCFT software.  Running 5.1.1.0 allows the avatar to function normally.

benaclejames, 2024-12-11:
My first recommendation would be to send logs

skunkmyrddyn, 2024-12-11:
[VRCFaceTracking.Services.ActivationService] Information: VRCFT Version 5.2.3.0 initializing...
[VRCFaceTracking.Services.ActivationService] Information: Initializing OSC...
[VRCFaceTracking.Core.OSC.OscQueryService] Debug: OSC Service Initializing
[VRCFaceTracking.Core.OSC.OscQueryService] Debug: OSC Service Initialized with result (False, False)
[VRCFaceTracking.Services.ActivationService] Information: Initializing main service...
[VRCFaceTracking.Core.Params.Data.UnifiedTrackingMutator] Debug: Reading configuration...
[VRCFaceTracking.Core.MainStandalone] Debug: Starting OSC update loop...
[VRCFaceTracking.Services.ActivationService] Debug: Checking for deletion requests for installed modules...
[VRCFaceTracking.Services.ActivationService] Information: Checking for updates for installed modules...
[VRCFaceTracking.Core.Params.Data.UnifiedTrackingMutator] Debug: Configuration loaded.
[VRCFaceTracking.Services.ActivationService] Information: Initializing modules...
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Starting initialization tracking
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Tearing down all modules...
[Microsoft.Extensions.Hosting.Internal.Host] Debug: Hosting starting
[VRCFaceTracking.Core.Library.UnifiedLibManager] Debug: SRanipalExtTrackingInterface properly implements ExtTrackingModule.
[VRCFaceTracking.Core.Library.UnifiedLibManager] Debug: Initializing requested runtimes...
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Initializing runtimes...
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Initializing SRanipalExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Loading External Module SRanipalExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[Microsoft.Hosting.Lifetime] Information: Application started. Press Ctrl+C to shut down.
[Microsoft.Hosting.Lifetime] Information: Hosting environment: Production
[Microsoft.Hosting.Lifetime] Information: Content root path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.2.3.0_x64__4s4k90pjvq32p\
[Microsoft.Extensions.Hosting.Internal.Host] Debug: Hosting started
[SRanipalExtTrackingInterface] Information: SRanipalExtTrackingModule: SRanipal version: 1.3.2.0
[SRanipalExtTrackingInterface] Information: Initializing Eye...
[SRanipalExtTrackingInterface] Information: Eye successfully started!
[SRanipalExtTrackingInterface] Information: Initializing Lip...
[SRanipalExtTrackingInterface] Information: Lip successfully started!
[VRCFaceTracking.Core.Library.UnifiedLibManager] Debug: Starting thread for SRanipalExtTrackingInterface
[VRCFaceTracking.Core.Library.UnifiedLibManager] Information: Tracking initialized via SRanipalExtTrackingInterface.SRanipalExtTrackingInterface

skunkmyrddyn, 2024-12-11:
Vive Pro Eye and the Vive Face tracker.

skunkmyrddyn, 2024-12-11:
No..  It happens immediately with the avatar.  I can do a OSC disable and enable in VRC and it will function for less than 1s, then the eyes will stop tracking.  Just the eye rotation, I should add.  Eye lid, and face never stop tracking.
Running 5.1.1.0, everything runs fine.

benaclejames, 2024-12-11:
Huh, does this issue persist regardless of the avatar?

skunkmyrddyn, 2024-12-11:
The legacy version of the same avatar does not exhibit this behavior, but I believe the creator updated the tracking system between the versions.

benaclejames, 2024-12-11:
Hmm. I wonder if this could be an avatar setup issue. Could you try some public known working avatars? There are quite a few in https://vrchat.com/home/world/wrld_b24fbb7c-9369-4cff-9242-32a35d44a8e8/info

skunkmyrddyn, 2024-12-11:
I can try a sample.  I've been in contact with the creator, and they're confused by it.  It was another user that suggested that the downgrade to 5.1.1.0 got it working.  But I figured I'd bring it up here since there seemed to be a need to use the previous version, and either help get it more accessible, or help figure out the why.

skunkmyrddyn, 2024-12-11:
I also have the avatar open currently in unity, if there's anything I could check on the setup that might explain it

benaclejames, 2024-12-11:
In general, a requirement to use an older version of the app is a flaw on the avatar creator's part. We only modify the way parameters are sent in compliance with our documentation and unified expressions spec, so avatars breaking due to updates are considered noncompliant

benaclejames, 2025-05-13:
Stale

closed 2025-05-13 by benaclejames