benaclejames/VRCFaceTracking #246: New Audio To Expressions Compatibility?

Rew7247, 2024-12-16:
Is there any way to get VRCFT to work with the new A2E update to Q3, Q3S, and Q2. I use the quest 3. I don't know if I would need to make my own plugin for it (Not that I think I can lol). Or if there is something I need to do for it to work.

Thank you and have a good rest of your day - Rew :3

regzo2, 2024-12-16:
They work with the existing Quest pro face tracking extensions, just install the appropriate Quest Pro compatible module for your setup (this isn't really documented anywhere, probably should update the VRCFT docs with this information).

Rew7247, 2024-12-17:
I tried but it doesn't send any data nothing happens.
I tried with airlink, regular link, Virtual Desktop, and steam link - with the respective modules, I'm thinking they might need something new enabled in the extensions since they got updated.

dfgHiatus, 2025-07-24:
Closing this issue due to inactivity.

closed 2025-07-24 by dfgHiatus