benaclejames/VRCFaceTracking #250: Auto Start Disabled

Odette8372, 2025-01-17:
[image]

After formatting my Windows 11, I reinstalled VRCFT, and the Auto Start setting is grayed out. I can't activate it. Does anyone have a solution?

AlexisEvo, 2025-01-18:
https://github.com/benaclejames/VRCFaceTracking/issues/243

Autostart was removed a while ago due to compatability issues. The above issue is tracking reintroducing it.

Odette8372, 2025-01-18:
@AlexisEvo 

Are you referring to the issue where Windows recognizes that SteamVR is on when VRCFT is on? I definitely had that problem before, so I had to manually turn off VRCFT when restarting SteamVR.

dfgHiatus, 2025-07-24:
Closing this as a duplicate of #243

closed 2025-07-24 by dfgHiatus