benaclejames/VRCFaceTracking #260: Vive XR Elite face tracking module - no eyebrow movement

puppykasa, 2025-02-13:
I recently bought the Vive XR Elite headset with its face tracking module and no matter what avatar I use, the eyebrows won't move. I did double check to make sure my avatars have the capabilities as I make them myself but also tried others. I came across a thread from 5 years ago with a similar issue that was fixed [here](https://forum.htc.com/topic/8493-no-way-to-track-eyebrows/)

This is my first time using github so I hope I did this right.

puppykasa, 2025-02-13:
[image]

AlexisEvo, 2025-02-14:
For stuff like this it's typically better to post on the Discord. The XR Elite (and all Vive headsets) don't have eyebrow support. iirc the XR Elite had eye widen at launch, but was broken in an update. The XRE has been getting updates as Vive tries to fix the Focus Vision, but I'm not sure if widen or emulated brow was on it. Can likely find it in the Viverse discord.

https://docs.vrcft.io/docs/hardware/interface-compatibilities

ps: death to wxs

puppykasa, 2025-02-14:
Yup.. I unfortunately found that out a bit after asking. Ordering an Oculus Pro rn 😭
What did wxs do to you??
Thank you, though!

closed 2025-05-13 by benaclejames