benaclejames/VRCFaceTracking #270: Eyes no longer tracked

Sydeqere2, 2025-04-10:
Meta Qust Pro no longer tracks eyes. The face is tracked, eye blinking occurs with a delay of 2 seconds

FacelessReaper, 2025-04-11:
Same issue, Quest pro, face working, eyes completely frozen.

kvalek, 2025-04-12:
Same issue. Using Meta Quest Link Module. Parameters for the eyes seem to be slow to arrive in VRchat. Like closing your eyes (when i eventually get them to close) will take seconds to open again after i RL open them. Eye look does not track at all. Have re-calibrated in the headset as well, but no luck.

Gatetor, 2025-04-13:
Same issue here; worked fine for 1 session (after a fully clean Windows installation on a new PC), then back to broken eyes

Zeddar, 2025-04-14:
Got the same issue. It does seem to track eye widening but not movement. I'm using the Quest pro link Module

Gatetor, 2025-04-14:
For everyone having this issue; the problem seems to be v76 of the Meta Link desktop app

Reverting to v74 fixes this issue; [here's the download for it](https://www.meta.com/en-gb/help/quest/1517439565442928/?srsltid=AfmBOop6d1Oxx3TnFwwKpgpV5SyM6NSWeUcrdUkhJ00RuCF6YEru3hb6)

[Here a more detailed guide on what to do](https://communityforums.atmeta.com/t5/Get-Help/Latest-Meta-Update-Breaks-eye-tracking-and-avatars-for-QUEST-PRO/td-p/1311837)

closed 2025-07-24 by dfgHiatus