benaclejames/VRCFaceTracking #271: I just can't install the program lol

iredyfox, 2025-04-14:
when opening the installer file, an error pops up and the install button glows gray.

text: By installing, you agree that this app will automatically check for and install updates. This app
package's publisher certificate could not be verified. Contact your system administrator or the
app developer to obtain a new app package with verified certificates. The root certificate and
all immediate certificates of the signature in the app package must be verified (0x800B010A)

plz fix this

song1710, 2025-04-16:
If the certificate expires, you can temporarily change the computer time to March 2025, and you can install it, and change it back after installation

SpecterArdy, 2025-04-18:
I got a version without the cert and it's standalone exe

benaclejames, 2025-05-13:
Closing due to [steam release](https://store.steampowered.com/app/3329480/VRCFaceTracking/)

closed 2025-05-13 by benaclejames