benaclejames/VRCFaceTracking #272: Rebuild without MSIX (No certificate just raw .exe)

SpecterArdy, 2025-04-18:
there is truly no reason to use .MSIX, as that is more intended for Microsoft apps (windows store), a native exe and a .msi installer package works better than .MSIX

i have done that with a clone of this repo on my profile and updated it's code to support .NET 9

if you're reading this and want the copy of a working build head to my profile, until the author updates it.

KaleidonKep99, 2025-04-18:
You could also just take the MSIX and unpack it, that's how I deal with it anyway.
I honestly also do not understand the need to repack it as a MSIX.

[image]

SpecterArdy, 2025-04-18:
the point of this is to skip .MSIX entirely and build a native app

benaclejames, 2025-05-13:
Closing due to [steam release](https://store.steampowered.com/app/3329480/VRCFaceTracking/)
We will be including raw .zip releases as part of new CI/CD going forward.

closed 2025-05-13 by benaclejames