benaclejames/VRCFaceTracking #280: Pico 4 Enterprise - Expose individual eye tracking for VRCFT

SapphireSouls, 2025-05-26:
Reposting from ALVR github https://github.com/alvr-org/ALVR/issues/2832

The PicoXR documentation for eye tracking seems to imply eye convergance (separate eye tracking) is supported on the Enterprise headset specifically. VRCFT currently claims this is an unsupported feature on their documentation.

https://developer.picoxr.com/document/unity/eye-tracking/

If you scroll down to the bottom of the page, you can see "GetPerEyePose" is referenced for Enterprise headsets only. Additionally, pupil data seems to be available too.

I don't know how the communication between VRCFT modules works, but I've raised this to ALVR too and just wanted to make sure I have all bases covered by making you aware of these features too, as your documentation seems to suggest otherwise.

I would love to see proper support for the Enterprise in VRCFT.

miranda1000, 2025-07-27:
@SapphireSouls it is theoretically supported, but only on Unity. PICO Connect API (at least, the old API) doesn't report it

miranda1000, 2025-07-27:
Also this should be posted on the PICO module repo, not the VRCFT one https://github.com/regzo2/PicoStreamingAssistantFTUDP

SapphireSouls, 2025-07-27:
> Also this should be posted on the PICO module repo, not the VRCFT one https://github.com/regzo2/PicoStreamingAssistantFTUDP

As a contributor on that repo I would assume you're aware of the request, but I'll repost it there for clarity.